numaru/vscode-ceedling-test-adapter #2: Test Explorer always reloading tests

thiagoperroni, 2018-10-02:
The UI stays on this state forever. The workspace has only a ceedling initial project with only one test. I got no problems running tests over the terminal.

[image]

numaru, 2018-10-04:
In which terminal do you use Ceedling?
For now the ext use 'cmd.exe' as default shell. Also Ceedling (and gcc by default) must be accessible in this shell to get the list of test (and execute them later).

thiagoperroni, 2018-10-07:
Sorry for the delay.

I use Ceedling on 'cmd.exe'. Using `shellPath` option as default (null),  the problem is the same I reported previously. 

Configuring `shellPath`  with the 'cmd.exe' path, I got this new error:

[image]

At least it has an error message now. I'll try some more things before give up.

Thanks for your attention

closed 2018-10-08 by numaru

numaru, 2018-10-08:
Thanks, i was able to reproduce it and patch it ✌. You should now be able to run from cmd.exe (`shellPath` not set). If one of you can confirm, I will let this issue closed.

aenima1891, 2018-10-09:
It works! Thank you!

thiagoperroni, 2018-10-09:
It worked for me too! Thank you so much!

commit 88ac56d9 (2018-10-08): :bug: Fix #2 by triming the paths gotten from cmd.exe

A carriage return char was causing a dead lock in fs.readFileSynch()
inside setTestSuiteInfo().