numaru/vscode-ceedling-test-adapter #9: Complains about not being able to run the project in non-ceedling projects

ejpcmac, 2018-10-08:
## Description

In non-C projects and C projets not using Ceedling, the following error shows on startup:

```
Ceedling failed to run the project. Please check the ceedlingExplorer.projectPath option.
```

AFAIK, this issue wasn’t present in 1.0.0 but is there since 1.1.0.

## Steps to reproduce

1. Install `vscode-ceedling-test-adapter`
2. Open a non-Ceedling project

numaru, 2018-10-10:
Yes, it is a side effect of warning the user if he failed to config his ceedling project. Just having an empty test window on first try doesn't make sense to me either.

A "disable warning" button (which set an option in the user config file) in the error panel could do the trick.

joel-felcana, 2019-07-03:
Hi,

Sorry for reopening a closed ticket, but I don't understand how this was fixed. I have a workspace with three folders (my source code in /src, some docs in /doc and the tests in /test). I keep them separated so I can get different c_cpp_settings.json for source and for tests, for example.
When I open the test explorer, I get two errors on the two workspaces that are not ceedling projects. How can I get rid of them?

[image]

I set the extension settings in the folder level, so only the test folder has the right path to the "project.yml" file (the other two have the default '.', and I think the extension is searching in `/src` and in `/docs` for a project.yml). As you said, it's better to get the error so the user can see that the problem comes from the extension is unable to get the settings, but maybe there should be a way to say "this folder doesn't contain one, that's why you can't find it, don't search".

numaru, 2019-07-03:
The error message was an annoying message box (in the right bottom corner) and It has been changed to an error node to avoid to spam the user on each project opening. The error message is now just less visible.

This does not fix the problem you are describing which, I think, deserve a separate issue.