numaru/vscode-ceedling-test-adapter #12: Error while debugging test: Error: Method not implemented.

aenima1891, 2018-10-12:
Hello,
I'm trying to debug a test, but I receive this message from VSCODE
[image]

I've ceedling, gcc, g++, gdb in my System PATH environment variable.

Thank you for the help

numaru, 2018-10-12:
Yes, the debug feature is not implemented for now 😉

aenima1891, 2018-10-13:
I'm debugging each test using task.json to compile the test (GCC) and launch.json to debug (gdb).
It's not the easiest way, but It works.
I Hope you'll implement this feature soon.
😁

numaru, 2019-07-03:
I'm closing this because the debug feature is now available (since [v1.2.0](https://github.com/numaru/vscode-ceedling-test-adapter/blob/master/CHANGELOG.md#120---2019-02-12))

closed 2019-07-03 by numaru