numaru/vscode-ceedling-test-adapter #13: Support differents project.yml configuration files

aenima1891, 2018-10-24:
I'd like to use different project.yml configuration files, becase I've a project with different configurations and every configuration has its own defines.

The Ceedling manual says:

> To use a project file name other than the default project.yml or place the project file in a directory other than the one in which you'll run Rake, create an environment variable CEEDLING_MAIN_PROJECT_FILE with your desired project file path.

So, i'm able to run different *.yml files by vscode terminal
`SET CEEDLING_MAIN_PROJECT_FILE=custom_project.yml && ruby vendor/ceedling/bin/ceedling`

Unfortunately, the Ceedling Test Adapter is not able to set this variable

Can you add the support to run different project.yml files?

For example, you can recognize every *project,yml in the root directory?

Thank you!

numaru, 2018-10-24:
It should be able to add a `ceedlingExplorer.shellEnv` option to let the user set its `CEEDLING_MAIN_PROJECT_FILE=custom_project.yml` or any other env variable.
Using multiple project.yml would be possible but it would require to edit the config between each swaps.

numaru, 2018-11-09:
If I understand right, the need is to be able to run multiple config of ceedling in the explorer tree.
If I transform the config to be like that (breaking change):

```js
{
    "ceedlingExplorer.configs": {
        "config_x": {
            "projectPath": "path/to/ceedling/project_z",
            "options": "options:variation_x"
        },
        "config_y": {
            "projectPath": "path/to/ceedling/project_z",
            "options": "options:variation_y"
        },
    },
    "ceedlingExplorer.shellPath": "path/to/shell/to/run"
}
```

and change the explorer tree to:

```
.
+---config_x
|   +---Test/Test_1
|   |   +---test_function_1_of_1
|   |
|   +---Test/Test_2
|       +---test_function_1_of_2
|
+---config_y
    +---Test/Test_1
    |   +---test_function_1_of_1
    |
    +---Test/Test_2
        +---test_function_1_of_2
```
Then both way of defining the multiple options should be possible.

Also, using `CEEDLING_MAIN_PROJECT_FILE` to use a different file name than `project.yml` is now out of scope of this issue.

sanko-oz, 2018-11-09:
This looks like a good solution. But keep in mind, that these individual configs (x/y) are actually a combination of the main config file and the config_x or config_y.  

In my specific case for example all I have in config_x is
```
:defines:
  :test:
    - _USER_DEFINE_
   - .....
```
this way my specific test file is compiled differently then the other test files. 
I actually like the idea of grouping the tests by config files. But it should be somehow obvious that these configs inherit from the main config file. 

And by the way. I really want to thank you for this awesome VS code extension. It makes life so much easier. Keep up the really good work.