numaru/vscode-ceedling-test-adapter #14: Support monorepo projects

ejpcmac, 2018-11-08:
While the configuration option `ceedlingExplorer.projectPath` adds support for a project root which is not the current workspace root, it does not handle workspaces with multiple subprojects.

For instance, let’s imagine the following layout:

```
*
|--A
|  |-- x
|  |-- project.yml
|
|--B
   |-- x
   |-- project.yml
```

Currently, I can do:

```json
{
    "ceedlingExplorer.projectPath": "A"
}
```

However, if I want to switch to project B, I need to update the project path. It would be great if I was able to do something like:

```json
{
    "ceedlingExplorer.projectPath": "A:B"
}
```

The path separator could be `:` as in the `PATH` variable, or `;`.

numaru, 2018-11-09:
This seems to be related to #13, let's move there.

ejpcmac, 2018-11-09:
I’ve read it, and maybe I didn’t understand properly, but I think it is not the same request: in #13, the author asks to be able to use a file that is not named `project.yml`, but only one at a time. Here I ask to be able to use several `project.yml` from different subdirectories.

numaru, 2018-11-09:
You are right. These issues are linked by the proposal of array in the config.
I'm grouping these to avoid changing the config twice.