numaru/vscode-ceedling-test-adapter #25: Remove error nodes on workspace folders which have no project.yml

numaru, 2019-07-03:
Hi,

Sorry for reopening a closed ticket, but I don't understand how this was fixed. I have a workspace with three folders (my source code in /src, some docs in /doc and the tests in /test). I keep them separated so I can get different c_cpp_settings.json for source and for tests, for example.
When I open the test explorer, I get two errors on the two workspaces that are not ceedling projects. How can I get rid of them?

[image]

I set the extension settings in the folder level, so only the test folder has the right path to the "project.yml" file (the other two have the default '.', and I think the extension is searching in `/src` and in `/docs` for a project.yml). As you said, it's better to get the error so the user can see that the problem comes from the extension is unable to get the settings, but maybe there should be a way to say "this folder doesn't contain one, that's why you can't find it, don't search".

_Originally posted by @joel-felcana in https://github.com/numaru/vscode-ceedling-test-adapter/issues/9#issuecomment-508171950_

numaru, 2019-07-03:
A proper solution could be a `ceedlingExplorer.hasProject` boolean option to tell the extension to search or not for a ceedling project in this workspace/folder.

joel-felcana, 2019-07-12:
That would be brilliant, thanks for considering my suggestion