numaru/vscode-ceedling-test-adapter #27: Not currently debugging a ceedling project

joel-felcana, 2019-07-12:
Hi,

Since a couple of days ago I'm getting this error on a project that I would sweat it was working before, with no changes whatsoever (apart from VS Code or other extensions updating).

I have all the settigns on their default values, except the debug taks, which points to the right debug task. It seems that the extension can't (or can but doesn't) translate the `${command:ceedlingExplorer.debugTestExecutable}` variable, and the program name passed to the debugger is just getting the folder path, but not the .out for this specific test. Replacing the variable with the harcoded file I want to debug works, as expected.

[image]

Any ideas? I set my settings at the folder level, because I work in a multifolder workspace, but settings are not being overriden by workspace or user settings. My default console can run ceedling and gcc, and the run test function works as expected.

bd2357, 2019-09-09:
Having the same issue, Projects with just one folder and the launch.json in the .vscode folder works but if I have a workspace file Get "Not currently Debugging a Ceedling test"  and the Dialog "launch: program with path up to the build\test\out  followed by does not exist

numaru, 2019-09-09:
I have difficulty in reproducing the issue. I'm using the following dir tree.
```bash
.
├── .vscode
│   └── launch.json
├── build
│   ├── artifacts
│   ├── gcov
│   ├── logs
│   ├── temp
│   └── test
├── ceedling
├── project.yml
├── src
...
│   ├── AdcHardware.c
│   └── AdcHardware.h
├── test
...
│   ├── TestAdcHardware.c
│   └── support
├── vendor
│   └── ceedling
└── ws
    └── sensor.code-workspace
```

with the following `sensor.code-workspace`

```json
{
	"folders": [
		{
			"path": ".."
		}
	],
	"settings": {
		"ceedlingExplorer.debugConfiguration": "ceedlingExplorer"
	}
}
```
@bd2357 Does the problem happen when you enter in debug by clicking on the debug button as described by Joel?

Where do you put the workspace file?

bd2357, 2019-09-10:
./unittest/named_devices/bishorn
├── .vscode
│   ├── launch.json
│   └── settings.json
├── bishorn.code-workspace
├── bishorn_tests
│   └── support
├── build
│   ├── artifacts
│   ├── logs
│   ├── temp
│   └── test
├── ceedling.cmd
└── project.yml
./unittest/common_tests
├── .vscode
│   ├── launch.json
│   └── settings.json
├── api
│   └── test_api.c
├── math
│   ├── test_fft.c
│   └── test_jtcomplex.c
├── readme.txt
├── serdes
│   ├── hss
│   ├── off_test_serdes_status_message.c
│   └── test_serdes_tx_control.c
├── support
│   ├── datatype.h
│   ├── device.h
│   └── unity_config.h
└── verification
    └── test_example5.c
./source
├── .vscode
│   └── settings.json
├── api
│   └── api.c
├── common
│   ├── assert.h
│   ├── common.mk
│   ├── debug.c
│   ├── debug.h

The workspace file: in the named_devices/bishorn folder on top
{
    "folders": [
        {
            "path": "."
        },
        {
            "path": "..\\..\\common_tests"
        },
        {
            "path": "..\\..\\..\\source"
        }
    ],
    "settings": {
        "ceedlingExplorer.debugConfiguration": "ceedlingExplorer"
    }
}

The Launch file, which is in .vscode folders in both the Workspace folder and common_tests

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ceedlingExplorer",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/test/out/${command:ceedlingExplorer.debugTestExecutable}",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "miDebuggerPath": "C:/Ruby25-x64/msys64/mingw64/bin/gdb.exe",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        }
    ]
}

It finds and runs the tests with the “Error while loading tests” errors for two of the paths as expected
When I click on the bug icon in the test Explorer next to a listed test I get a status dialog in lower right:
Not Currently debugging a Ceedling Test
And a Visual Studio Code Dialog Window
[cid:image001.png@01D567B5.0CA180B0]

Thanks again for your help, Love Ceedling and your test runner is a nice addition, being able to debug these workspace file controlled projects would make t great.
I did mention it all works fine if I am just using a folder based project.
In my current project, the workspace file points at two locations with ceedling tests and one location that is the main source tree.  The project.yml file is in the same folder as the .code-workspace file but that is not where the current tests are being found. They are in the common_tests path. Kind of a mess but I inherited a very odd project tree and intend to have multiple .code-workspace files with multiple project.yml files across several folders.

Begin forwarded message:

***** Attention - External Email *****

numaru, 2019-09-10:
Thank you both. I was able to reproduce the problem. it should be fixed in the next release :relieved:

PR #37 (2019-09-10): The vscode multi dir projects cannot debug tests
A VSCode workspace with more than one directory cannot debug a test
using the command ceedlingExplorer.debugTestExecutable. A pop-up
appear containing the path to the debug executable without the last
part.

This commit fix the bug by using a global variable, common to every
CeedlingAdapter, to store the name of the currently debugged
executable.

fix #27
commit 26d4017d (2019-09-10): :bug: The vscode multi dir projects cannot debug tests

A VSCode workspace with more than one directory cannot debug a test
using the command ceedlingExplorer.debugTestExecutable. A pop-up
appear containing the path to the debug executable without the last
part.

This commit fix the bug by using a global variable, common to every
CeedlingAdapter, to store the name of the currently debugged
executable.

fix #27
closed 2019-09-10 by numaru