numaru/vscode-ceedling-test-adapter #32: xml_reports plugin is required but does not work

UG-Steven, 2019-08-12:
Basically, when I enable the xml_reports plugin I run into this issue:

`ceedling test:all 
C:/Ruby25/lib/ruby/2.5.0/rubygems/core_ext/kernel_require.rb:59:in `require':
 cannot load such file -- C:/Undagrid/repositories/stm_undabedded/ceedling/vendor/Ceedling/vendor/cmock/lib/cmock_generator_plugin_xml_tests_report.rb `

I tried resolving this issue here: [http://www.throwtheswitch.org/forums?place=msg%2Fthrowtheswitch%2F68DPra-CpGM%2F3y1oPLfYBgAJ](url). Since then I've been in contact with Mark from Ceedling. He is aware of the issue and it should be resolved with the next release.

Meanwhile, this plugin is still required to run the ceedling test adapter for visual studio and I'm seeing other people post issues I did not run into before. So I assume other people here have found a workaround or I am somehow on the wrong version of Ceedling. I have tried using the latest release of ceedling as well as the up to date code on the master branch. Nothing works. Is there an older version of Ceedling I need to use? I tried v0.28.2 but ran into the same issue. With older versions I'm running into different issues. Should I try to fix those issues so I can try the even older versions? Maybe there's simply another option I'm missing that will fix it? Any help is greatly appreciated.

numaru, 2019-08-12:
Hello,

I use the `temp_sensor` example project to do my tests.
```bash
# Create a temp_sensor project example
ceedling example temp_sensor
# ...
# Open the dir
cd temp_sensor
# Open project.yml in vscode to enable `xml_tests_report`
#
# :plugins:
#  :load_paths:
#    - vendor/ceedling/plugins
#  :enabled:
#    - xml_tests_report
#    - stdout_pretty_tests_report
#    - module_generator
#    - gcov
code project.yml
# Run the tests
ceedling
# ...
# Open the generated report.xml
code build/artifacts/test/report.xml
```
in `build/test/cache/input.yml`, there is the following entry `:xml_tests_report_path: vendor/ceedling/plugins/xml_tests_report`.

**version**
```bash
Laboratory/Local/temp_sensor > ceedling version
  Ceedling:: 0.28.3
CException:: 1.3.1.18
     CMock:: 2.4.6.217
     Unity:: 2.4.3.122

Laboratory/Local/temp_sensor > ruby --version
ruby 2.6.3p62 (2019-04-16 revision 67580) [x86_64-linux]
```
I never encounter the issue you describe.

Despite this issue is not related to this repo, I let it open a bit in case anyone already found a workaround and want to share it here.

UG-Steven, 2019-08-15:
Thanks for the help! I figured out the problem, I made a mistake while enabling the xml reports plugin, I put it under cmock plugins 🤦‍♀️

numaru, 2019-08-15:
Ok, no prob. I'm closing this.

closed 2019-08-15 by numaru