numaru/vscode-ceedling-test-adapter #39: Ceedling failed to run in the configured shell

Zoey-SunflowerLabs, 2019-10-01:
I am using ceedling in WSL. So, I configured "ceedlingExplorer.shellPath": "C:/WINDOWS/System32/wsl.exe". But it showed "Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option." Did I configure it wrong?

numaru, 2019-10-01:
wsl doesn't support the `-c` nor `/d /s /c` argument. You may use `C:\Windows\System32\bash.exe` instead.

Unfortunately the current nodejs version embedded in vscode doesn't use the right arguments when calling bash on windows. This has been fixed in nodejs/node@af883e1f99598492474c13447e2a1895c0c6f1b7 so you can expect this to be fixed on update of nodejs to version >= v11.0.0.

The update to electron v6 (and nodejs > v12) was planned for the September release but it seems delayed to October one. 

In the meantime, you should still be able to use the `vscode-ceedling-test-adapter` through the [remote-wsl](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-wsl) extension (even with the `ceedlingExplorer.shellPath` unset).

the7i11, 2019-10-28:
I'm currently facing an issue with similar symptoms, but instead of trying to run ceedling inside WSL, the error appears with `ceedlingExplorer.shellPath: null` on Windows 7.
I can call ceedling from both PowerShell and cmd.exe. 
Do you have any idea why that is?

numaru, 2019-10-28:
@the7i11 Can you please open a separate issue mentioning the windows 7 os. On w10 with both `shellPath: null` and `shellPath: "null"`, I'm able to discover the tests.

numaru, 2019-11-14:
The vscode october update bumped electron to v6.
I'm closing this issue. The wsl can now be used as shell without the remote-wsl by using `"ceedlingExplorer.shellPath": "bash"`.

closed 2019-11-14 by numaru