numaru/vscode-ceedling-test-adapter #40: Windows 7: Ceedling failed to run in the configured Shell

the7i11, 2019-10-29:
I'm using the ceedling test adapter on Windows 7.
No matter what I tried to configure the shell to, the only output I get is `Ceedling failed to run in the configured Shell`. I tried the following values for "ceedlingExplorer.shellPath: "
- `null`
- `"null"`
- `cmd.exe`
- `C:\Windows\System32\cmd.exe`
- `C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe`

numaru, 2020-03-23:
Is it still relevant? I don't have any Windows 7 on hand to test this but I remember trying it once. If someone passing by can let a word or an emoji, it would be much appreciated 💌

jnz86, 2023-12-05:
WIndows 7 was EOL in 2020, and Extended Support was removed on Jan 10 2023. 

Seems like this could be closed for housekeeping.

closed 2023-12-07 by the7i11