numaru/vscode-ceedling-test-adapter #46: Windows 10: Ceedling failed to run in the configured Shell

kbsteere, 2020-03-20:
I have attempted to include the bash.exe, powershell.exe and all paths to those as well but I still get the error message: "Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option."

numaru, 2020-03-20:
Can you run the ceedling command line in those shells ?

kbsteere, 2020-03-20:
Yes, I can run it with cmd and powershell without any issue.

numaru, 2020-03-20:
Is this message prompted when you clicked the error in the tree view?

[image]

If yes, I think the issue is that the extension require the `xml_test_report` plugin to parse the results. Also, a line needs to be added in the `project.yml`

```diff
:plugins:
  :load_paths:
    - vendor/ceedling/plugins
  :enabled:
+   - xml_tests_report
    - stdout_pretty_tests_report
    - module_generator
    - gcov
```

numaru, 2020-03-21:
Also, what is your ceedling version?

I tried with these ones on w10:

```powershell
C:\Users\kin\Documents\Laboratory\github\win32\vscode-ceedling-test-adapter [master ≡]
> ceedling version
Welcome to Ceedling!
  Ceedling::   0.28.3
  CMock::      2.4.5
  Unity::      2.4.2
  CException:: 1.3.1
C:\Users\kin\Documents\Laboratory\github\win32\vscode-ceedling-test-adapter [master ≡]
> (get-command ceedling).path
C:\Ruby24-x64\bin\ceedling.bat
```
with no `ceedlingExplorer.shellPath` defined.

kbsteere, 2020-03-23:
I never see the error about xml_reports

Here are my versions
VS Code : 1.43.1
Electron: 7.1.11
Node.js: 12.8.1

Welcome to Ceedling!
    Ceedling:: 0.29.1
       CMock:: 2.5.1
       Unity:: 2.5.0
  CException:: 1.3.2

numaru, 2020-03-23:
I updated my Ceedling version and I'm still be able to use the extension. 🐱 

To find out if the issue come from the environment or the Ceedling project, can you create a new project using the following command `ceedling example temp_sensor`? I'm using this project to test the extension. I assume we are testing with the lastest ceedling and vscode versions, using no extension settings.

FYI: The extension shows the error message if it fails to run the command `ceedling summary` in the `ceedlingExplorer.projectPath`.

## e.g. 

```powershell
C:\Users\kin\Documents\Laboratory\github\win32\temp_sensor
> ceedling summary ; write-host $? ; write-host $env:comspec

--------------------
OVERALL TEST SUMMARY
--------------------
TESTED:  47
PASSED:  47
FAILED:   0
IGNORED:  0

--------------------------
GCOV: OVERALL TEST SUMMARY
--------------------------

No tests executed.

NOTE: Summaries may be out of date with project sources.

True
C:\WINDOWS\system32\cmd.exe
```

mb81, 2020-05-01:
I can do this: 
[image]

But i am not able to get the plugin running due to **"Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option."** Is there a way to get this running using wsl?
Ceedling:: 0.29.1
Unity:: 2.5.0
CMock:: 2.5.1
CException:: 1.3.2

numaru, 2020-05-01:
@mb81 Despite the fact the error message is the same, I see the problem you are describing as a separated one because this is not the same shell used. I think it deserve its own issue.

Anyway, I tried with:

* The temp_sensor example generated with `ceedling example temp_sensor` and the `xml_report` plugin added in `project.yml`.
* The following `.vscode/settings.json`
```json
{
    "ceedlingExplorer.shellPath": "bash"
}
```

and it works.

A workaround could be to use the wsl extension of vscode. see #39

mb81, 2020-05-02:
Thank you for letting me trying again the "bash" in shellPath, which i tried several times, and in different variants. **It works ^^ .** 
Also it works with the remote Wsl connection.  
Nice tool :)