numaru/vscode-ceedling-test-adapter #48: How to configure ceedling command line

thomasfinchbr, 2020-04-09:
Hi,

First of all thanks for the great plugin.
I would like to know how I can configure ceedling plugin in vs code to run the command line
ceedling gcov:all

I understand that it always runs the option ceedling test:all.

Thanks.

numaru, 2020-04-14:
Yes you are right. It is not possible to run `ceedling gcov` in the extension for now. Is there a benefit to run `gcov` instead of `test`?

thomasfinchbr, 2020-04-17:
I run the option 'ceedling gcov:all utils:gcov' to get code coverage of the unit tests. 
Whit this option the output is a little different.
I get a xml with the code coverage that can but used with another vs code plugin.
I think that it is not possible to the coverage with the option 'test:all'

Thanks.

numaru, 2020-04-20:
✨ It's now implemented as a `ceedlingExplorer.testCommandArgs` option. I share a config in the PR to use [Coverage Gutters](https://marketplace.visualstudio.com/items?itemName=ryanluker.vscode-coverage-gutters).

Thank you for the contribution. 😺

closed 2020-04-20 by numaru

thomasfinchbr, 2020-04-22:
Thanks. I will test it!

PR #51 (2020-04-20): Add  a testCommandArgs option to be able to run gcov
It should fix #48.

---

I have been able to use the [Coverare Gutters](https://marketplace.visualstudio.com/items?itemName=ryanluker.vscode-coverage-gutters) extension on Windows.

**settings.json**
```json
{
    "ceedlingExplorer.testCommandArgs": [
        "test:${TEST_ID}",
        "gcov:${TEST_ID}",
        "utils:gcov"
    ],
    "coverage-gutters.xmlname": "GcovCoverageResults.xml",
    "coverage-gutters.coverageReportFileName": "**/GcovCoverageResults.html",
}
```

and after installing gcovr as described in the [ceedling documentatio [... cut]