numaru/vscode-ceedling-test-adapter #49: Using a script for shell path

GrantRolls, 2020-04-13:
I am utilising a docker environment for development. 
I currently am using a script to run things inside that docker container. My test tasks run the following command

`run-in-docker.ps1 nrf ceedling`
[image]

I have tried setting the shell path as
` "ceedlingExplorer.shellPath": "\"C:\\Program Files\\PowerShell\\7\\pwsh.exe\" -Command run-in-docker.ps1 nrf"`
And this works from a command prompt, and if I append `ceedling` it runs ceedling as expected

But I get an error when using the test explorer
`Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option.`

numaru, 2020-04-14:
The extension tries to give `-c "ceedling summary"` to the unknown shells. To be compatible, the powershell script should accept `"C:\\Program Files\\PowerShell\\7\\pwsh.exe\" -Command run-in-docker.ps1 nrf -c "ceedling summary"`.

Here is a minimal example of a "fake shell" which can be used by the extension.

**fake-shell.bat:**
```bat
call cmd /c %2
```

**settings.json**
```js
{
  "ceedlingExplorer.shellPath": "fake-shell.bat"
}
```

RiMi-Vx, 2023-04-14:
Hi numaru,

It would be nice when it was possible to set a relative path to the "shell". 
We would like to put the batch file right beside the ceedling "project.yaml" within our SCM repo.

I've tried to adapt your source by simply using the same approach for the "shellPath" like you already used for the "projectPath".

What do you think about such a change/enhancement?
 
[image]

numaru, 2023-04-15:
The path to shellPath is either absolute or relative to the current directory.

[image]

I'm not sure to understand why you would need it to be relative to the projectPath.

RiMi-Vx, 2023-04-17:
Thank you for the clarification.  I did not get that the "shellPath" ist relative to the "projectPath" 👍

closed 2024-01-18 by GrantRolls