numaru/vscode-ceedling-test-adapter #53: Multiline test function names are not picked up

sanko-oz, 2020-04-21:
Since my test function names are starting to get very long, I started to use multiple lines like so:

void test_\
GIVEN_xxxxxxxxxx__\
WHEN_xxxxxxxxxxxxxx__\
THEN_xxxxxxxxxxxxxxxxxx(void)
{
}

Ceedling is not having any problems with this notation, but the test-adapter is unfortunately not able to find the test functions in that case.

numaru, 2020-04-29:
Thank you for the report. I'm closing the issue because it is fixed by the PR.

closed 2020-04-29 by numaru

sanko-oz, 2020-05-11:
Unfortunately its not working for me. I updated to 1.5.1 but my test functions are still not getting picked up.

numaru, 2020-05-11:
You are right, the current published version does not support the feature. I bet I got myself in a complete muddle when I published it.

I'm publishing the right file with the same version number 1.5.1. You might want to uninstall/reinstall the extension to get the latest version.

sanko-oz, 2020-05-12:
Thank you very much. Now its working as expected.