numaru/vscode-ceedling-test-adapter #66: Extension does not work on Mac catalina, vscodium, oh-my-zsh shell

krish2487, 2020-09-12:
Hello,
I have installed and use this configuration on mac catalina with vscodium and oh-my-zsh shell.
No matter what the value of
`"ceedlingExplorer.shellPath":` is set to, the test explorer is not discovering any tests.
I have tried with various combinations of zsh, sh, bash, null with and without double quotes.

Ceedling is installed as a gem and is available on the system path. In fact, I tried it on an existing project to test the extension.

And yes, the xml_reports plugin is enabled in the plugins section of the project.yml.

I would be grateful if anyone can help me fix this issue. I am guessing I am doing something wrong.. but am unable to figure out where.

numaru, 2020-09-13:
What I did to try to reproduce the issue:
- install vscodium inside a macOS 10.15 vm using `brew cask install vscodium`.
- install ceedling using `sudo gem install ceedling`
- generate a temp_sensor project using `ceedling example temp_sensor`
- install the `vscode-ceedling-test-explorer` from the .vsix I download on the store website because I could not find it in the in app marketplace.
- open the temp_sensor project using `code temp_sensor`
- add the `xml_tests_report` to the `:plugins` section of project.yml
- reload the window and it works.
- also, my `settings.json` is empty.

[image]

Could you try to generate the temp_sensor on your machine too ?

krish2487, 2020-09-13:
Hello @numaru . Thank you for the rapid response.
I am able to replicate your findings with the temp_sensor project.

On a hunch, I deleted the contents of settings.json in my original project folder,  after reading the last point you have mentioned. And the plugin now works even in this project. I am able to see the tests and the plugin doesnt throw an error message.

Thank you @numaru . You have been very helpful.
I am closing this issue now.

krish2487, 2020-09-13:
Uhh.. Looks like I spoke to soon.
I restarted my machine to continue and the same issue seems to reappear again.

Both in my project and the temp_sensor project. 
I cant see the tests anymore and It throws the same "Ceedling : command not found"
[image]
[image]

@numaru  Any other information or logs I can provide to make troubleshooting easier ?

Thank you again so much and I apologize for reopening a rather petty. I am almost certain its something I must be doing wrong.

krish2487, 2020-09-14:
Okay. 
I have had some more time to experiment and found a more or less repeatable way to reproduce the issue, and maybe a potential solution.

The test explorer **always** fails to discover any tests when I start vscodium from the spotlight or the dock, irrespective of the project.

However running `code .` or `code <project_folder_name>` to open the project files results in the vscodium editor results in the test explorer working just fine and the tests run as well.

I am not sure if this is intentional or a corner case, for now I will use this workflow and continue ahead.
@numaru  Thanks again for the help. :-) 
I ll close the issue again for now. :-P

closed 2020-09-14 by krish2487