numaru/vscode-ceedling-test-adapter #67: Test explorer not working with parameterized tests

LeMimit, 2020-09-15:
Hello ! 

I have the following test:

```c
/**
 * @brief Check that we handle incorrect values
 */
TEST_CASE( "0x123", 10)
TEST_CASE( "abc", 10)
void test_convertStrToSignedNumber_invalid_input(char *input, uint8_t base)
{
  long value;
  enum app_ecode_e rc;

  rc = convertStrToSignedNumber(input, &value, base);
  TEST_ASSERT_EQUAL( rc, APP_ECODE_BAD_PARAMETER );

}
```

When I am trying to run it, the status is not updated in the test explorer interface. The test is a success in the report.

[image]

**I do not have any problems with tests which are not parameterized.**

[image]

**Seems like parameterized tests is not supported by the extensions ? May be I need to add some options ?**

Thanks a lot !

numaru, 2020-09-15:
The parameterized tests are not yet available in the extension.

The way the extention works currently:
* The extension parse all the files given by `ceedling:test` to get a list of test using a regex (`^((?:\\s*TEST_CASE\\s*\\(.*?\\)\\s*)*)\\s*void\\s+((?:${testPrefix})(?:.*\\\\\\s+)*.*)\\s*\\(\\s*(.*)\\s*\\)`) but nothing is done with the TEST_CASE. The list of test is used to build the tree view.
* When the run button is pressed, `ceedling test:${testFile}` is run and the `report.xml` is read to get the results.

I don't think there is a way to get a list of all the tests from the ceedling command line without running them. This is why the extension use a regex.

In the `report.xml`, the parameterized tests are named with a suffix. For example, `test/TestFoo.c::test_convertStrToSignedNumber_invalid_input("0x123", 10)`.

To support the parameterized test in the extension, I think we should add an entry in the tree view for each new params. I'm not sure if it should be done after the test run using the `report.xml` (need to check if it can be done as a test adapter) or with the regex (what happen if someone use another macro than `TEST_CASE` as it is in this workaround? https://github.com/ThrowTheSwitch/Ceedling/issues/241, we should make this configurable).

LeMimit, 2020-09-16:
Ok, thanks for the answer.

Seems like it should be more intuitive to have the parameterized test appears after the discovery of the tests, so it should be done with the regex.

Do you plan to add this feature in the near futur ?

If no, may be I can help you on this ? How do you mange this ? Can you give me some more information on how modify the architecture to add this feature ?

Thanks !

fj-sanchez, 2021-02-08:
It would be great to get this feature.

numaru, 2021-02-15:
I'm looking at this

LeMimit, 2021-02-15:
Thanks for adding this feature! 👍🏻

fj-sanchez, 2021-02-16:
Thanks @numaru for adding it, I'll also try it tomorrow and provide feedback.

clabough, 2021-03-22:
Will this also make it so that this shows pass/fail for each test instead of a gray circle with square inside?

```
  describe.each([['0'], ['-1']])('should invalidate integer length values less than 1', badValue => {
    it(` "${badValue}"`, () => {
      expect(component.validNumber(badValue)).toBe(false);
    });
  });
```

numaru, 2021-03-23:
@clabough Sorry, I’m afraid I don’t follow you. I don't understand what you would expect and when. And why your example it written in javascript? This repo handle the [ceedling](http://www.throwtheswitch.org/ceedling) adaption to the [VSCode test explorer](https://marketplace.visualstudio.com/items?itemName=hbenl.vscode-test-explorer) extension.

clabough, 2021-03-23:
@numaru Sorry, I ended up here somehow looking into an issue with Jest Test Explorer/Test Explorer UI. My mistake.

olowo726, 2022-03-21:
This seems to be implemented but since the issue isn't closed I'll write a feature request here.

We have to use the workaround described at https://github.com/ThrowTheSwitch/Ceedling/issues/195. There is a solution to avoid this workaround (https://github.com/ThrowTheSwitch/Ceedling/pull/497) but in involves compiler-specific flags for modifying macro processing behaviour. This workaround breaks our existing test cases.

Hence, could I please have an extension setting enabling me to tell the extension to use TEST_VALUE (or arbitrary string) instead of TEST_CASE? The same applies to TEST_RANGE.

numaru, 2022-04-02:
> This seems to be implemented but since the issue isn't closed I'll write a feature request here.
> 
> We have to use the workaround described at [ThrowTheSwitch/Ceedling#195](https://github.com/ThrowTheSwitch/Ceedling/issues/195). There is a solution to avoid this workaround ([ThrowTheSwitch/Ceedling#497](https://github.com/ThrowTheSwitch/Ceedling/pull/497)) but in involves compiler-specific flags for modifying macro processing behaviour. This workaround breaks our existing test cases.
> 
> Hence, could I please have an extension setting enabling me to tell the extension to use TEST_VALUE (or arbitrary string) instead of TEST_CASE? The same applies to TEST_RANGE.

new issue opened here https://github.com/numaru/vscode-ceedling-test-adapter/issues/91

closed 2022-04-02 by numaru