numaru/vscode-ceedling-test-adapter #68: Linux, Ceedling failed to run in the configured shell, how to troubleshoot?

ykoehler, 2020-11-24:
I have set the ceedlingExplorer.shellPath to point to the proper location (somewhere where ceedling binary is located, one level before).

Yet I am still getting this message

Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option.

I am not that knowledgeable about VSCode yet as to understand how I can troubleshoot what is happening.  I am using VScode on Windows with a Remote environment using SSH on Linux CentOS7

In the Terminal plugin within vscode, I can confirm my paths are correct, that the binary is working fine.  I can use my test script to run the ceedling test as well.  The VScode ceeding plugin is the only place where things fall apart.

numaru, 2020-12-07:
Do you use the [ms-vscode-remote.remote-ssh](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-ssh) plugin? We faced the same issue with the WSL and the remote plugin was a workaround.

Also, what is your ceedlingExplorer.shellPath option?

numaru, 2020-12-07:
Unfortunately, there is no debug log nor unit tests for this extension. Here is how I debug:
- First, I try to reproduce the problem with the vscode window opened by pressing F5 in the extension repository.
- Then, I put breakpoints in the code and watch the memory until I find what's wrong.

hpe-ykoehler, 2022-12-14:
I think the issue is that this is expected to be a shell program like bash... where I thought it was going to be the path to the Ceedling binary...

Aptahar, 2024-05-16:
I have the same issue, output:  
`2024-05-16 14:17:42.701 [error] Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option.`
Try `bash`  and `null` and `bin/bash` in ceedlingExplorer.shellPath, same error.  
OS Ubuntu.

hpe-ykoehler, 2024-05-16:
have you try "/bin/bash"? and validated that bash is in that /bin folder?  The missing / at the beginning of the string may be why it is not working for you.

Aptahar, 2024-05-16:
> have you try "/bin/bash"? and validated that bash is in that /bin folder? The missing / at the beginning of the string may be why it is not working for you.

Yes, even tried exec `/bin/bash -c "ceedling summary"` in terminal in project folder and it work

chintanp, 2025-02-20:
@numaru this problem still exists.