numaru/vscode-ceedling-test-adapter #72: ShellPath issue: please help me figure it out

embcla, 2021-02-10:
Hello

First of all thank you for making this extension, I was just trying to make a build configuration and was going crazy.
Happy to see someone made a complete extension!!! 👏 

My environment is the following:
Operating System: Win10
toolchain: `C:\Qt\Qt5.12.10\5.12.10\mingw73_32\bin`
shell: `git-bash`
ceedling: via rake binaries in path
```
     Ceedling:: 0.30.0
        CMock:: 2.5.2
          Unity:: 2.5.1
 CException:: 1.3.2
```

so when I'm in my gitbash I can just do the `ceeling test:mytest`
VSCode is loaded with the same global environment

I saw another issue and created a _call-ceedling.bat_ with this content
`call git-bash.exe /c %2`

and put this into my vs
`"ceedlingExplorer.shellPath": "C:/Program Files/Git/call-ceedling.bat"`

And yet it says it can't run and there is an issue with the shell.
What am I missing here?

numaru, 2021-02-12:
Hello, I see you are using the `git-bash` with ceedling but I don't get how you installed ruby in it. There is no package manager in my git-bash. I bet you installed somewhere else but I cannot reproduce a similar environment to investigate the issue. 

Anyway, I think the issue come from the missing quotes around the path containing space in the `.bat` file.

[image]

You should use instead:

```
call "C:\Program Files\Git\bin\bash.exe" /c %2
```

numaru, 2021-02-12:
But I think that using the bash.exe path directly in the `ceedlingExplorer.shellPath` should work.

[image]

It is not working on my machine because ceedling is not installed in my git-bash but the command is launched.

embcla, 2021-02-12:
Thanks for looking into this.
I did put the path to git bash directly but it still fails.

The way I have ceedling running in git is via env vars.

1. Install Ruby (mine is at `C:\Ruby27-x64\bin\ruby.exe`)
2. Add Ruby bin folder to path
3. Install Ceedling (`gem install ceedling`)

Now you should have ceedling available in git bash

numaru, 2021-02-12:
Thanks, I added ruby to the path and I have now access to it in the git-bash.

In the .bat, the bash.exe option to execute a command is `-c` instead of `/c`.

```sh
C:\Users\kin\Documents\Laboratory\github\wsl\temp_sensor>fake-shell.bat "ceedling test"

C:\Users\kin\Documents\Laboratory\github\wsl\temp_sensor>call "C:\Program Files\Git\bin\bash.exe" -c "ceedling test"
ERROR: Config filepath [:tools][:gcov_compiler][:executable]['gcc'] does not exist in system search paths.
ERROR: Config filepath [:tools][:gcov_linker][:executable]['gcc'] does not exist in system search paths.
ERROR: Config filepath [:tools][:gcov_report][:executable]['gcov'] does not exist in system search paths.
ERROR: Config filepath [:tools][:test_compiler][:executable]['gcc.exe'] does not exist in system search paths.
ERROR: Config filepath [:tools][:test_file_preprocessor][:executable]['gcc.exe'] does not exist in system search paths.
ERROR: Config filepath [:tools][:test_includes_preprocessor][:executable]['gcc.exe'] does not exist in system search paths.
ERROR: Config filepath [:tools][:test_linker][:executable]['gcc.exe'] does not exist in system search paths.
C:/Users/kin/Documents/Laboratory/github/wsl/temp_sensor/vendor/ceedling/lib/ceedling/configurator.rb:290:in `validate': unhandled exception
        from C:/Users/kin/Documents/Laboratory/github/wsl/temp_sensor/vendor/ceedling/lib/ceedling/setupinator.rb:32:in `do_setup'
        from vendor/ceedling/lib/../lib/ceedling/rakefile.rb:48:in `<top (required)>'
        from vendor/ceedling/lib/ceedling.rb:66:in `load'
        from vendor/ceedling/lib/ceedling.rb:66:in `load_project'
        from C:/Ruby24-x64/lib/ruby/gems/2.4.0/gems/ceedling-0.29.1/bin/ceedling:318:in `<top (required)>'
        from C:/Ruby24-x64/bin/ceedling:23:in `load'
        from C:/Ruby24-x64/bin/ceedling:23:in `<main>'
```

It is still not working well on my machine but ceedling is found. Can you please, share a screenshot of your `OUTPUT` panel in vscode. I have the same error in vscode with both the fake-shell and the raw git-bash. I put my fake-shell.bat next to the project.yml file.

embcla, 2021-02-12:
[image]

[image]

Here you go

numaru, 2021-02-12:
I think you're putting your config in the right place.

It seems like there is an issue with a path containing space in the `ceedlingExplorer.shellPath` if bat file is used.

wrong:
[image]

right:
[image]

You still got no message under the first line of the OUTPUT if you put the `.bat` somewhere else ?

embcla, 2021-02-12:
I have to change the bat script to %1, otherwise it expects ceedling to be the 2nd command rather than the 1st, if I do that then I can manually call it

[image]

but regardless of this, the extension doesn't load

[image]

embcla, 2021-02-12:
Ok I finally managed!
I was setting my projectPath wrong: didn't read the docs well enough, and was putting ${workspaceRoot} in there, but it's already relative.

Don't know why it was giving me the message about the shellPath though, but happy that I'm running now!!!

embcla, 2021-02-12:
Is it normal behaviour that when I click on a unit test top element, it closes the entire terminal/output/overall/console tab?
when I click on one of the included tests, it reopens it and shows the output of the entire unit

numaru, 2021-02-12:
Nice !

Something is wrong with the error message. It is printed when ceedling cannot be executed but it should not lead us to the wrong place 😏 

I think the click behaviour is normal. It's handled by a lower layer, the https://github.com/hbenl/vscode-test-explorer extension.

closed 2021-02-12 by numaru

embcla, 2021-02-15:
Hi, I tried moving the batch file to the project folder, so I can make it part of the repo, but I'm back to the same issue.

[image]

numaru, 2021-02-15:
The `${workdspaceForlder}` is not expanded by the extension. Just use `"ceedlingExplorer.shellPath": "call-ceedling.bat"`, I think the path is relative to the workspace directory. But since the issue was in the `ceedlingExplorer.projectPath`, I'm pretty sure you could get rid of the `call-ceedling.bat` file now and use directly `"ceedlingExplorer.shellPath": "C:/Program Files/Git/bin/bash.exe"`.

The extension does not depends on the OS directly. It is abstracted by vscode/node api. Why would you need to do something different based on the OS?

embcla, 2021-02-15:
Well if I open the project in Windows then my shellPath would be a windows EXE with a windows path.
On the other hand, opening the same project in Linux, the shell would be a /bin/bash-zsh-sh etc

numaru, 2021-02-15:
There nothing in the extension to handle this use case. There is a pretty old issue (2016) in vscode which is still open. [microsoft/vscode#5595](https://bit.ly/3qrmTgI).
To address, this use case, I would not put any setting which is linked to the developer machine in the git repository and let him configure the extension. I would describe how to do it in a doc. (the REAMDE.md for example).