numaru/vscode-ceedling-test-adapter #77: Project path is not resolved correctly in version 1.8.0

kab4rt, 2021-02-17:
Hi,

Starting my work today I was surprised to find that the adapter stopped detecting my Ceedling project correctly with standard message: _Failed to find the project.yml file. Please check the ceedlingExplorer.projectPath option_. The important factor is that my project.yml file is not present in root workspace directory.

As this seemed a bit odd I have tried to find the root cause of the problem. Downgrading version to 1.7.0 solves the issue. In my opinion the problem is caused by this commit: https://github.com/numaru/vscode-ceedling-test-adapter/commit/8a55e09f72da6b00283c0d9dd2529439cd0ce94c , especially the line 241:
```js
 if (!(fs.existsSync(absolutePath) && !fs.lstatSync(absolutePath).isDirectory())) {
```

I guess what you tried to achieve is to enter fallback path in case specified path does not exist or points to a file instead of a directory. However, in my case the expression evaluates to _true_ and function takes fallback `defaultProjectPath` path. I guess that this change should solve the problem:
```js
if (!fs.existsSync(absolutePath) || !fs.lstatSync(absolutePath).isDirectory()) {
```

numaru, 2021-02-17:
Hello! I have updated the extension earlier this week. I might have broken something. I will try to have a look a soon as possible.

closed 2021-02-17 by numaru

numaru, 2021-02-17:
Thank you for the issue report and the time you took to find the root cause. It should be all fixed now. 😇

kab4rt, 2021-02-17:
Thank you as well, I can confirm version 1.8.1 is working as expected 👍

commit 3992a813 (2021-02-17): :bug: Fix the "always use default projectPath" issue

The issue happen even if the given directory exists.

fix #77