numaru/vscode-ceedling-test-adapter #80: Failed to find the project.yml file. Please check the ceedlingExplorer.projectPath option.

AdrianWenger1988, 2021-02-24:
Hi,

I cannot run the tests in Visual Code:
Followin error message appears:

Failed to find the project.yml file. Please check the ceedlingExplorer.projectPath option.

[image]

But there actually is a project.yml file in the current dir:

[image]

AdrianWenger1988, 2021-02-24:
The problem is repsoducable.
My colleague faces the same issue!

numaru, 2021-02-24:
Hello, it is working on my windows machine and I don't see obvious reason to fail in the code. But this message can happen also if the file exists but it can't load the content. I suspect a wrong project.yml file encoding. The expected file encoding is UTF-8. The encoding error message is not logged but you can try to use the extension with a clean project to see if it fixes the issue.

Usually, I use `ceedling example temp_sensor` to generate a project. Then I open the project.yml with vscode (make sure the file encoding is UTF-8 in the right bottom corner) and add `xml_tests_report` to the `plugin.enabled` section. I configure the `ceedlingExplorer.shellPath` option if ceedling is not accessible in my default shell and reload the tests.

AdrianWenger1988, 2021-02-24:
Thanks for your reply.
It is not the enconding of the project.yml, but there is an issue with the project.yml file content of the  seetings for gcov:

Settings which lead to the issue:
[image]

It works with this settings:
[image]

numaru, 2021-02-24:
Ok, perfect. I can close this. 👌

closed 2021-02-24 by numaru