numaru/vscode-ceedling-test-adapter #81: Load report xml from gcov directory

NovaNekmit, 2021-04-19:
(Code-Reference: https://github.com/numaru/vscode-ceedling-test-adapter/blob/master/src/adapter.ts#L692)

Currently the report xml is always loaded from the `artifacts/test/` directory, but doing `ceedling gcov:<test>` writes the report to `artifacts/gcov/` instead. So one must run both `gcov:${TEST_ID}` (to get the coverage) and `test:${TEST_ID}` (so this adapter can find the xml).

This means running the tests twice, when the report.xml could instead just be read from the gcov directory instead.

numaru, 2021-04-21:
👍 I forgot about that workaround (ref #51). I think we could take the last updated file. We would save up a setting option and stay gcov agnostic.

closed 2021-04-21 by numaru

numaru, 2021-04-21:
it should be fixed in the 1.8.2

NovaNekmit, 2021-04-21:
Works fine for me.

Unless ofcourse I tell ceedling to fail on missing coverage, ie via
```yml
:gcov:
  :abort_on_uncovered: TRUE
  :gcovr:
    :fail_under_line: 100
    :fail_under_branch: 100
```

numaru, 2021-04-21:
Yep mb, the issue is fixed and the messages are not the cause of the red warnings. I just forgot to reload my extension in vscode after the update 😅 

Thank you for the contribution ☺️

PR #83 (2021-04-21): Hotfix/1.8.2
Use the last udpated report.xml file as XmlReportPath because the file
is generated in a `gcov` subfolder when `ceedling gcov:*` is used.

fix #81
commit f9dbb8bc (2021-04-21): :bug: Return the right report.xml when using gcov

Use the last udpated report.xml file as XmlReportPath because the file
is generated in a `gcov` subfolder when `ceedling gcov:*` is used.

fix #81