numaru/vscode-ceedling-test-adapter #87: Migrating to native testing

AlexandreMarkus, 2022-01-14:
In version 1.59, VS Code added an official API and UI for running tests. Use it instead of Test Explorer UI, which is now deprecated

numaru, 2022-01-15:
I think the Test Explorer UI also added for us a translation layer. And `testExplorer.useNativeTesting: true` in the settings.json should enable the native experience.

I didn't use vscode testing in a year, is there any reason other than the deprecation to switch to the native API and remove the Test Explorer UI dep? (any missing feature or Quality of Life improvement?)

AlexandreMarkus, 2022-01-17:
Removing one deprecated dependency will also help to load the extension faster. Both Ceedling Test Explorer and Test Explorer UI are quite slow to activate when opening VS Code, so it will improve the booting time if you remove the dependency.
Thank you for your great work!

AlmazShai, 2022-06-10:
Ceedling Test Explorer depends on the Test Explorer UI extension, which depends on the Test Adapter Converter. Thus, the need to install all of these extensions is not as clear to the user, and for testing and debugging common projects, a manual for installing all three extensions should be described. Does migrating to the native API make this task easier?