numaru/vscode-ceedling-test-adapter #91: Be able to use TEST_VALUE instead of TEST_CASE (https://github.com/ThrowTheSwitch/Ceedling/issues/195)

olowo726, 2022-03-24:
We have to use the workaround described at https://github.com/ThrowTheSwitch/Ceedling/issues/195. There is a solution to avoid this workaround (https://github.com/ThrowTheSwitch/Ceedling/pull/497) but in involves compiler-specific flags for modifying macro processing behaviour. This workaround breaks our existing test cases.

Hence, could I please have an extension setting enabling me to tell the extension to use TEST_VALUE (or arbitrary string) instead of TEST_CASE? The same applies to TEST_RANGE.

numaru, 2022-04-02:
I remember that someone might need to change the name of the macros as a workaround. 😋 Does the config proposed here could do the trick for you? https://github.com/ThrowTheSwitch/Ceedling/issues/241#issuecomment-775413905

olowo726, 2022-04-02:
> workaround. 😋 Does the config proposed here could do the trick for you? ThrowTheSwitch/Ceedling#241
> (comment)

This did not work for me since the compiler flag changes the processing of
other macros we're using in our tests. Hence I could not used it but have
to revert to the method of using TEST_VALUE instead of TEST_CASE. Hence I'm
asking to be able to use TEST_VALUE (or arbitrary string from a setting)
instead of TEST_CASE. I think this could be done perfectly backwards
compatible by having the default value of the string being TEST_CASE.

numaru, 2022-04-03:
Oh my bad, I missed the fact that they both use the same workaround :facepalm:. Ok I can look at it

olowo726, 2022-04-03:
Great, thanks!

commit 810e79e5 (2022-04-03): :sparkles: Add configurable aliases for TEST_CASE and TEST_RANGE

fix #91
closed 2022-04-03 by numaru