numaru/vscode-ceedling-test-adapter #98: Cannot run debug gdb

doubleell, 2022-04-07:
Hello,
I have followed steps to run Ceedling tests using VS Code. The tests run ok, however when I click the debug icon I have the following messages;
[image]

I have setup the debug launch etc, I have used in the past no problem and pointed to ceedling_gdb.

        {
            "name": "ceedling_gdb",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/ceedling/build/test/out/${command:ceedlingExplorer.debugTestExecutable}",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb",
            "miDebuggerPath": "C:/msys64/mingw32/bin/gdb.exe",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        },

numaru, 2022-04-07:
Hello, thank you for the report. A new release has been published this weekend but nothing should have changed in the "debug part". I tried to debug in the wsl today and did not encounter any issue.

Can you reproduce it with the v1.8.2?

[image]

doubleell, 2022-04-08:
Hi,
Thanks for the reply. I tried v1.8.2 but that was the same. I can’t see
what the problem is, strange….

doubleell, 2022-04-08:
Hi,
I have just found the issue. I had another folder open in the workspace
that doesn't have any ceedling yaml file. I think it was somehow trying to
run tests for this project instead even though I had clicked the icon for
the correct debug, the run icon for the test worked ok;

[image: image.png]

Regards,
Allan.

numaru, 2022-04-10:
Sure! This need be to be investigated

FTchips, 2022-04-15:
Hello,
I am also running into this issue. I try to debug a test by hitting the debug icon in the test explorer, and nothing happens and the same three exceptions pop up.
I had fixed it by specifying the folder in the [workspaceFolder variable](https://code.visualstudio.com/docs/editor/variables-reference#_variables-scoped-per-workspace-folder) in the debug launch: 
```
{
    "name": "Ceedling Test Explorer Debug",
    "type": "cppdbg",
    "request": "launch",
    "program": "${workspaceFolder:**MyFolder**}/build/test/out/${command:ceedlingExplorer.debugTestExecutable}",
    "args": [],
    "stopAtEntry": false,
    "cwd": "${workspaceFolder:**MyFolder**}",
    "environment": [],
    "externalConsole": false,
    "MIMode": "gdb",
    "miDebuggerPath": "C:\\Ruby27-x64\\msys64\\mingw64\\bin\\gdb.exe",
    "setupCommands": [
        {
            "description": "Enable pretty-printing for gdb",
            "text": "-enable-pretty-printing",
            "ignoreFailures": true
        }
    ]
}
```
This worked and I was able to debug tests in all folders. 

However, after adding another folder to the workspace, it seems to have broken again. I have added the debug launch to the new folder with  correct scope, and added the launch name to the debugConfiguration, added the project.yml but still encounter the errors. 
I can run tests from all folders in the workspace, and see statuses for all folders. 
But, I am unable to debug any tests from any of the folders.

Is there a log that might provide more information about where things are going wrong?

numaru, 2022-04-15:
It should be fixed after #99

closed 2022-04-15 by numaru

numaru, 2022-04-15:
> Is there a log that might provide more information about where things are going wrong?

Nop, the extension is lacking of this kind of feature. Currently I use the debugger to follow the execution path :smile: 

Thank you both for the report! Feel free to reopen if you still encounter the issue

commit 49973a6c (2022-04-15): :bug: Look through all the adapters for a debug executable

The adapter factory is called on every workspace folders. Then there is
multiple instances of the adapter initiliazed when the debug function is
called. But only one of them has a debugTestExecutable set when the
debugTestExecutable command is called. We can use this to choose which
executable name to return.

This commit also fix the clean and clobber commands for workspaces with
multiple folders. Now the commands are executed on all the adapter
instances instead of the last initialized one.

fix #98