numaru/vscode-ceedling-test-adapter #101: Using instructions, When I open the test view I get an error

Jerryc4174, 2022-05-03:
I followed the instruction on your web post and fot the following error when I opened the test view
[image]
"error command test-explorer.reload-collection.  Command "test-explorer.reload-collection"  not found.

Jerryc4174, 2022-05-04:
Kin – I appreciate the response.  Here are the versions installed: Vscode Test Explorer v2.21.1Vscode v1.66.2Ceedling Test Adapter 1.9.1 Regards Jerry Crane Sent from Mail for Windows From: Kin NumaruSent: Wednesday, May 4, 2022 1:30 PMTo: numaru/vscode-ceedling-test-adapterCc: Jerryc4174; AuthorSubject: Re: [numaru/vscode-ceedling-test-adapter] Using instructions, When I open the test view I get an error (Issue #101) Do you have a link to this web post or maybe you are talking about the README?This extension does not provide the test-explorer.reload-collection command. It is just a bit of glue to run ceedling commands and parse the results. It relies and depends on another extension https://github.com/hbenl/vscode-test-explorer which provide the incriminated command. But I have never seen this message before.Could you share the version of vscode-test-explorer, vscode-ceedling-test-adapter and vscode you are using? Maybe with that, I should be able to reproduce the issue.

Jerryc4174, 2022-05-15:
Kin – Just an update.  I was able to get the test adapter working by installing mingw32 using msys2.  It is now working perfectly.  Sorry to raise the issue. Jerry Crane Sent from Mail for Windows From: Kin NumaruSent: Wednesday, May 4, 2022 1:30 PMTo: numaru/vscode-ceedling-test-adapterCc: Jerryc4174; AuthorSubject: Re: [numaru/vscode-ceedling-test-adapter] Using instructions, When I open the test view I get an error (Issue #101) Do you have a link to this web post or maybe you are talking about the README?This extension does not provide the test-explorer.reload-collection command. It is just a bit of glue to run ceedling commands and parse the results. It relies and depends on another extension https://github.com/hbenl/vscode-test-explorer which provide the incriminated command. But I have never seen this message before.Could you share the version of vscode-test-explorer, vscode-ceedling-test-adapter and vscode you are using? Maybe with that, I should be able to reproduce the issue.

numaru, 2022-06-04:
Thank you for the follow up 😃

closed 2022-06-04 by numaru