numaru/vscode-ceedling-test-adapter #102: Coloured output is not handled

lipi, 2022-06-01:
When enabling Unity's coloured_report plugin, which turns pass/fail results green/red, the test explorer's output doesn't handle the control characters:

[image]

Expected behaviour:  at least it should remove the control characters from the output.  Ideally it would colour the output properly. 

Note, that VSCode's terminal does handle the colours properly:

[image]

numaru, 2022-06-04:
👍 Thank you for the clear report. I will have a look!

numaru, 2022-06-04:
After a quick investigation. The text is given to the output panel GUI using a `vscode.OutputChannel`. And adding the ascii color support to the native output panel has been removed from the vscode roadmap. But you can use for the [IBM/output-colorizer]( https://marketplace.visualstudio.com/items?itemName=IBM.output-colorizer) extension to add ascii color to the output.

What I can do is: remove the escape codes by default and add a settings option to re-enable them if you use a third-party extension like the IBM one.

lipi, 2022-06-04:
I have tried the IBM plugin but it seems to add extra coloring for certain keywords, it does not improve on the original issue.

numaru, 2022-06-04:
Oups, my bad :facepalm: I tested the extension really quick and I lost myself. But, we can now remove the escape codes in any case :smile:

closed 2022-06-04 by numaru

numaru, 2022-06-04:
It should be now fixed in the new 1.10.0 release

PR #103 (2022-06-04): :sparkles: Add an option to remove the ascii escape sequences
The escape sequences are not evaluated by the OUTPUT panel. We should
remove them. I let an option to re-enable them if one day it's somehow
supported by vscode.

fix #102
commit 51c4899a (2022-06-04): :sparkles: Add an option to remove the ascii escape sequences

The escape sequences are not evaluated by the OUTPUT panel. We should
remove them. I let an option to re-enable them if one day it's somehow
supported by vscode.

fix #102