numaru/vscode-ceedling-test-adapter #105: Tests do not load after upgrade to 1.10.0

tkaier, 2022-06-07:
In the Side panel, no tests show up with the version v1.10.0

This is the case for native Windows and WSL. After downgrading the Ceedling test adapter plugin to v1.9.1 the tests reappear.

closed 2022-06-07 by numaru

numaru, 2022-06-07:
Thank you for the report! 

:facepalm: I confess! I changed some import syntax in the release branch and did not even test on my machine...

[image]

py4mac, 2022-06-08:
Thanks for the update.

tkaier, 2022-06-09:
Thank you for the update.

I love your plugin and use it on a daily basis. And I am impressed by the fast response time to this issue. So many thanks for your work.

commit a933d90a (2022-06-07): :bug: Correctly import async-mutex using ES6 syntax

fix #105