numaru/vscode-ceedling-test-adapter #107: UNRESPONSIVE extension host: 'numaru.vscode-ceedling-test-adapter' when using ceedlingExplorer.testCaseMacroAliases

olowo726, 2022-06-09:
Hi,

For version 1.10.1 I experience 

`[warning] UNRESPONSIVE extension host: 'numaru.vscode-ceedling-test-adapter' took 97% of 5184.417ms, saved PROFILE here: 'file:///tmp/exthost-f33d44.cpuprofile' [{"id":"numaru.vscode-ceedling-test-adapter","total":5005607,"percentage":97},{"id":"program","total":178571,"percentage":3}]`

when 

```
"ceedlingExplorer.testCaseMacroAliases": [
        "TEST_VALUE"
      ]
```
It doesn't get  stuck for any other value for testCaseMacroAliases but then the only alias used in our code is TEST_VALUE

olowo726, 2022-06-21:
This would be very useful for me so just let me know if I can help in any way.

numaru, 2022-06-21:
Hello, I tried to reproduce without success. I suspect the function regex but I would like to reproduce to be able to fix it. With 100 files which contains 32 tests each, I'm still below 150ms to parse the functions.

Is your project open source? If not, could you make a "minimal project" which has the issue and share it?

olowo726, 2022-06-22:
Hi,

Unfortunately the project isn't open source but I managed to reduce it to quite a small project that I can share with the issue, see attached
[namura.zip](https://github.com/numaru/vscode-ceedling-test-adapter/files/8958508/namura.zip)
.

numaru, 2022-06-22:
Is there the issue without any .c files? I have a "Ceedling: Error" node in the test view but not the described unresponsive message

olowo726, 2022-06-22:
Yea, it's there even when I remove all source files.

numaru, 2022-06-22:
Arg, it's gonna be a tough one :smirk_cat: I tried to install the other extensions used in settings.json but still no message. Should I expect a pop up or a message in the terminal?
Does the issue only happen with `"TEST_VALUE"` when no source is here also?

numaru, 2022-06-22:
Can you share the generated `.cpuprofile` file?