numaru/vscode-ceedling-test-adapter #109: Remote Extension host terminating unexpectedly when trying to use ceedling test adapter

gabrielcobostello, 2022-07-15:
Hi!

So i recently downloaded ceedling (0.31.1) and wanted to give it a try with vscode 1.69.1.

I was using the temperature sensor example from ceedling to test out the tool and the ceedling test adapter, but for some reason the testing tap just shows Ceedling: Error. The only information I get is that there was an error loading tests and when I click on it I dont get any more info on the error.

The Test Explorer: Reloading Tests is constantly running and eventually the pannel reloads itself and I get a message stating **"Remote Extension host terminated unexpectedly"**.

The thing is that I normally work in a **remote server to which i connect to by ssh**, and I develop all my code and tests there. I dont have linux locally so I can not test if this problem araises from the remote connection alone or if I am doing something wrong, but im quite sure I ran all the setup steps correctly, as shown in the extension page.

Let me know if I can provide any more information.

Best regards,
Gabriel.

numaru, 2022-07-17:
Thank you for the report,

Can you run the tests using the ceedling CLI? Is the `Test Explorer UI` and the `Ceedling Test Explorer` extensions installed in the remote vscode?

Is it the same thing with the version 1.9.1 of the extension? 
[image]

The "infinite loading with no error" reminds me this other ticket #107

gabrielcobostello, 2022-07-18:
Ok so I realiced the issue with the constant termination was due to having the Unity test explorer extension, so that is solved now (I removed it).

The Ceedling extension still fails but now i can better see the error (I also get the same error with 1.9.1):
```
Ceedling failed to run in the configured shell. Please check the ceedlingExplorer.shellPath option.
ERROR: Ceedling Failed

rake aborted!
Don't know how to build task 'ceedling summary' (see --tasks)
/var/lib/gems/2.5.0/gems/ceedling-0.31.1/bin/ceedling:345:in `block in <top (required)>'
/var/lib/gems/2.5.0/gems/ceedling-0.31.1/bin/ceedling:332:in `<top (required)>'
/usr/local/bin/ceedling:23:in `load'
/usr/local/bin/ceedling:23:in `<main>'
(See full trace by running task with --trace)
```

When I do run `ceedling summary --trace` on the terminal it seems to work:
```
--------------------
OVERALL TEST SUMMARY
--------------------
TESTED:  47
PASSED:  47
FAILED:   0
IGNORED:  0

--------------------------
GCOV: OVERALL TEST SUMMARY
--------------------------

No tests executed.

NOTE: Summaries may be out of date with project sources.
```

My setting are configured to point to the installation directory of ceedling as follows:
[image]

So im not sure what is going on.

The terminal seems to work perfectly with ceedling, I can run all tests no problem. And I have the extensions installed in both the local and remote PCs.

Any help is appreciated.

numaru, 2022-07-18:
Ho nice catch! 👁️ I didn't know about the collision with the other extension.

For the `shellPath` setting, it is the path to the shell to use when the extension run the ceedling commands. I think you could let it empty or set to `null`, it should then use `sh` or `bash` by default. But if you need some env variables defined in your `.bashrc` for example, you can set it explicitly to `/bin/bash`... This setting is more useful on windows to give the path to a mingw exe where ceedling is installed.

gabrielcobostello, 2022-07-19:
I only modified the `shellPath` because of the message when ceedling failed, but yeah it didnt make much sense to specify a path when it is already on my `bash` .

If I set it to `null` it works! So thx for the help :) It was just confusion with the other unity package causing interfearance I guess.