numaru/vscode-ceedling-test-adapter #111: Unsupported option leads to misleading "Unable to find project file"

embcla, 2023-03-17:
Hi everyone

Someone is using the following ceedling option in their project.yml
```
 :gcov:
     :html_report_type: basic
```
When that's enabled, test explorer throws the infamous 
`Failed to find the project.yml file. Please check the ceedlingExplorer.projectPath option.`

VSCODE
Version: 1.76.2
Commit: ee2b180d582a7f601fa6ecfdad8d9fd269ab1884
Date: 2023-03-14T17:57:21.103Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Linux x64 5.15.0-67-generic snap
Sandboxed: No

Ceedling Test Explorer: 1.10.1

breakersun, 2023-07-07:
I don't have gcov enabled but still have same problem with ceedling temp_sensor example

<details>
<summary>Click me</summary>

```yaml
---

# Notes:
# Sample project C code is not presently written to produce a release artifact.
# As such, release build options are disabled.
# This sample, therefore, only demonstrates running a collection of unit tests.

:project:
  :use_test_preprocessor: TRUE
  :use_auxiliary_dependencies: TRUE
  :build_root: build
  # :release_build: TRUE
  :test_file_prefix: Test
  :which_ceedling: gem
  :ceedling_version: '?'

#:release_build:
#  :output: TempSensor.out
#  :use_assembly: FALSE

:environment: []

:extension:
  :executable: .out

:paths:
  :test:
    - +:test/**
    - -:test/support
  :source:
    - src/**
  :support:
    - test/support

:defines:
  :common: &common_defines []
  :test:
    - *common_defines
    - TEST
  :test_preprocess:
    - *common_defines
    - TEST

:cmock:
  :when_no_prototypes: :warn
  :enforce_strict_ordering: TRUE
  :plugins:
    - :ignore
  :treat_as:
    uint8:    HEX8
    uint16:   HEX16
    uint32:   UINT32
    int8:     INT8
    bool:     UINT8

:libraries:
  :system:
    - m

:plugins:
  :load_paths:
    - "#{Ceedling.load_path}"
  :enabled:
    - stdout_pretty_tests_report
    - module_generator
...

```

</details>

numaru, 2023-11-29:
This error can happen if the yaml file is invalid or if the file is not found by the extension.

@breakersun the yaml file is valid, it's maybe the other reason. I tried locally and it is working.
@embcla I tried to add this option to a `project.yml` file and I cannot reproduce the issue. Maybe I could with your full file.

I'm closing the issue because I cannot reproduce. Feel free to reopen if you have more info.