numaru/vscode-ceedling-test-adapter #117: does VSCode extension support "options:<custom_yaml_file" ?

Pierre-Denis, 2023-09-15:
Hello, 

I was wandering if your extension currently support the practical arg "[options](https://github.com/ThrowTheSwitch/Ceedling/blob/master/docs/CeedlingPacket.md)" offered by Ceedling that allow to merge/load custom configuration to an existing one for a given launch (combination of both your main yaml and a custom one for a given test). 

Thank you in advance. 

Bye

numaru, 2023-09-16:
Nop, it's not supported :disappointed:. For most of the features, the extension acts as a wrapper around the ceedling CLI. But to get some of the info, the `project.yml` file is parsed and none of the `option_paths` is followed. To be honest I didn't even knew this feature existed :sweat_smile:.

Pierre-Denis, 2023-09-18:
Ok thank you very much for your answer. 
Hope this will be considered in the future as it is very practical to test multiple configuration of the same code or Hardware Abstraction Layer based on preprocessing (eg: DEBUG, RELEASE, HAL CHIP MODEL).
Regards

JuPrgn, 2023-12-19:
This feature would be very appreciated, I have just discovered the extension but my projects are based on a `project.yml` and `project_specific.yml` file. 

Calling `ceedling summary` report :

> ERROR: Config filepath [:tools][:test_compiler][:executable]['gcc.exe'] does not exist in system search paths.
> ERROR: Config filepath [:tools][:test_dependencies_generator][:executable]['gcc.exe'] does not exist in system search paths.
> ERROR: Config filepath [:tools][:test_file_preprocessor][:executable]['gcc.exe'] does not exist in system search paths.
> ERROR: Config filepath [:tools][:test_file_preprocessor_directives][:executable]['gcc.exe'] does not exist in system search paths.
> ERROR: Config filepath [:tools][:test_includes_preprocessor][:executable]['gcc.exe'] does not exist in system search paths.
> ERROR: Config filepath [:tools][:test_linker][:executable]['gcc.exe'] does not exist in system search paths.

but calling `ceedling project:project_specific.yml summary` is working fine as compiler path are specified in `specific.yml`

JuPrgn, 2023-12-19:
Just modified `adapter.js` to add `project:project_specific.yml ` to every `ceedling ` cmd

```
    getCeedlingCommand(args) {
        const line = `ceedling project:project_specific.yml ${args.join(" ")}`;
        return line;
    }
```
I don't know how VS extension works but I will try to propose a PR to add a setting for this.

MikeExMachina, 2024-03-21:
Just wanted to chime in that I would love to see support for the "options" feature as well. I usually have multiple HALs for my project to support different hardware builds, options are great for appending the paths of a specific HAL for inclusion in a build.