numaru/vscode-ceedling-test-adapter #118: Ceedling 0.32.0 will change the path of the test binary output

swaldhoer, 2023-10-16:
The output directory structure of Ceedling will change:
 
| Version    | Output path                                                                                        |
|------------|----------------------------------------------------------------------------------------------------|
| ``0.31.1`` | ``./test/out/${command:ceedlingExplorer.debugTestExecutable}``                                     |
| ``0.32.0`` | ``./test/out/${test_file_name_without_extension}/${command:ceedlingExplorer.debugTestExecutable}`` |
 
This means, if you have a test ``test_my_test_abc.c`` the ``debugTestExecutable``
path changes to
 
``./test/out/test_my_test_abc/${command:ceedlingExplorer.debugTestExecutable}``
 
This is a crucial update for us, although I have no TypeScript/JavaScript experience, if you could point me to the code where this needs to be fixed, I would try to prepare a Pull request.

cwilliamson-bti, 2023-11-17:
I also have no TypeScript/JavaScript experience or experience writing VSCode extensions. Looking through the source I think the solution is to just add a command in the main.ts and then some changes in the adapter.ts.
Again no idea what I'm doing but as a temporary hack I changed the files in extension directly located in `C:\Users\<user>\.vscode\extensions\numaru.vscode-ceedling-test-adapter-1.10.1\out\src` to the attached files, problemMatcher.js didn't need to be changed.
[src.zip](https://github.com/numaru/vscode-ceedling-test-adapter/files/13394558/src.zip)
From there I edited the launch.json program line to:
`"program": "${workspaceFolder}/tools/ceedling/build/test/out/${command:ceedlingExplorer.testFileName}/${command:ceedlingExplorer.debugTestExecutable}",`

This is an extreme hack but I have no idea how to edit the source code to produce a vscode extension. There must be some "compile" process that takes the typescript source in the repo and converts it to javascript. I live in embedded C so this is quite a different environment to me. Anyway hope this helps you and I'm sure anyone who knows what they're doing could quickly submit a pull request with the fix.

swaldhoer, 2023-11-18:
@cwilliamson-bti Does ``${command:ceedlingExplorer.testFileName}`` for a file ``test_abc.c`` resolve to

- ``test_abc``

or to

- ``test_abc.c``?

swaldhoer, 2023-11-18:
Based on this code snippet

https://github.com/numaru/vscode-ceedling-test-adapter/blob/420dc8b6d65df007616cf56ca056542e71c26ae0/src/adapter.ts#L172-L178

I assume, that ``${command:ceedlingExplorer.testFileName}`` resolves to ``test_abc``.

Then the solution to this issue is to just set the workspace specific path correctly, i.e., to something like this:

```json
{
    "name": "Ceedling Test Explorer Debug",
    "type": "cppdbg",
    "request": "launch",
    "program": "${workspaceFolder}/build/test/out/${command:ceedlingExplorer.testFileName}/${command:ceedlingExplorer.debugTestExecutable}",
    "args": [],
    "stopAtEntry": false,
    "cwd": "${workspaceFolder}",
    "environment": [],
    "externalConsole": false,
    "MIMode": "gdb",
    "miDebuggerPath": "C:/MinGW/bin/gdb.exe",
    "setupCommands": [
        {
            "description": "Enable pretty-printing for gdb",
            "text": "-enable-pretty-printing",
            "ignoreFailures": true
        }
    ]
}
```

swaldhoer, 2023-11-24:
The setting described above works so I am closing this issue.

PR #129 (2024-01-02): :sparkles: Add debug compatibility for ceedling >= v0.32.0
The path to the debug executable was computed depending on some `project.yml` options. It's not a big deal to depend also on the ceedling version.

fix: #118
duplicate: #124
commit 02f066aa (2024-01-02): :sparkles: Add debug compatibility for ceedling >= v0.32.0

The path to the debug executable was computed depending on some
`project.yml` options. It's not a big deal to depend also on the
ceedling version.

fix: #118
duplicate: #124
closed 2023-11-24 by swaldhoer