numaru/vscode-ceedling-test-adapter #120: Ceedling Test Explorer Play Button No Longer Running Test

lexstarwind, 2023-11-07:
I have a simple Unity / Ceedling / CMock example project from this tutorial (https://embetronicx.com/tutorials/unit_testing/unit-testing-in-c-testing-with-unity/). Nothing special. I was able to run this project using the Ceeding Test Explorer for VSCode on my previous windows 10 machine. I've since switched to a Windows 11 machine and the same project won't run when pressing the play button in the test explorer window. Strangely enough the tests all run from command line. They also run in the VSCode terminal if I press the debug button, but the tests will not run in the test explorer window if I press the play button. Any ideas?

I get the following error
Errno::EINVAL: Invalid argument @ rb_file_s_mtime

project.yml

---

# Notes:
# Sample project C code is not presently written to produce a release artifact.
# As such, release build options are disabled.
# This sample, therefore, only demonstrates running a collection of unit tests.

:project:
  :use_exceptions: FALSE
  :use_test_preprocessor: TRUE
  :use_auxiliary_dependencies: TRUE
  :build_root: build
#  :release_build: TRUE
  :test_file_prefix: test_
  :which_ceedling: gem
  :ceedling_version: 0.31.1
  :default_tasks:
    - test:all

#:test_build:
#  :use_assembly: TRUE

#:release_build:
#  :output: MyApp.out
#  :use_assembly: FALSE

:environment:

:extension:
  :executable: .out

:paths:
  :test:
    - +:test/**
    - -:test/support
  :source:
    - src/**
  :support:
    - test/support
  :libraries: []

:defines:
  # in order to add common defines:
  #  1) remove the trailing [] from the :common: section
  #  2) add entries to the :common: section (e.g. :test: has TEST defined)
  :common: &common_defines []
  :test:
    - *common_defines
    - TEST
  :test_preprocess:
    - *common_defines
    - TEST

:cmock:
  :mock_prefix: mock_
  :when_no_prototypes: :warn
  :enforce_strict_ordering: TRUE
  :plugins:
    - :ignore
    - :callback
    - :array
  :treat_as:
    uint8:    HEX8
    uint16:   HEX16
    uint32:   UINT32
    int8:     INT8
    bool:     UINT8

# Add -gcov to the plugins list to make sure of the gcov plugin
# You will need to have gcov and gcovr both installed to make it work.
# For more information on these options, see docs in plugins/gcov
:gcov:
  :reports:
    - HtmlDetailed
  :gcovr:
    :html_medium_threshold: 75
    :html_high_threshold: 90

#:tools:
# Ceedling defaults to using gcc for compiling, linking, etc.
# As [:tools] is blank, gcc will be used (so long as it's in your system path)
# See documentation to configure a given toolchain for use

# LIBRARIES
# These libraries are automatically injected into the build process. Those specified as
# common will be used in all types of builds. Otherwise, libraries can be injected in just
# tests or releases. These options are MERGED with the options in supplemental yaml files.
:libraries:
  :placement: :end
  :flag: "-l${1}"
  :path_flag: "-L ${1}"
  :system: []    # for example, you might list 'm' to grab the math library
  :test: []
  :release: []

:plugins:
  :load_paths:
    - "#{Ceedling.load_path}"
  :enabled:
    - stdout_pretty_tests_report
    - module_generator
    - gcov
    - xml_tests_report
...

miebka, 2023-11-14:
I also stumbled across the same problem during a fresh installation on a Win10 machine. 
It seems that `${TEST_ID}` from the call to ceedling contains the path to the unit test module.

`Errno::EINVAL: Invalid argument @ rb_file_s_mtime - test:test/test_helloWorld.c`

```json
{
    "ceedlingExplorer.testCommandArgs": [        
        "test:${TEST_ID}"
    ]
}
```

It is a strange behavior, the old installations of my colleagues work without any problems, even with new VSCode versions. But if you uninstall Ruby (3.0.6-1)  and Ceedling (0.31.1) and reinstall them, you run into this problem.

lexstarwind, 2023-11-14:
Thanks for the reply. I'll give that a shot. After talking to a coworker, I believe it may be a security issue. My company recently changed IT companies so people are having similar issues with various development environments.

lexstarwind, 2023-11-16:
Ok so after further testing, it is not a security issue. 
I still get the same error. Errno::EINVAL: Invalid argument @ rb_file_s_mtime
I'm not sure why. The tests run fine from the command line, just not when I press play using the test runner.

lexstarwind, 2023-11-16:
Full error output
No tests executed.

stderr:
rake aborted!
Errno::EINVAL: Invalid argument @ rb_file_s_mtime - test:test/test_bit_manipulation.c
C:/Users/lexstarwind/.local/share/gem/ruby/3.0.0/gems/ceedling-0.31.1/bin/ceedling:345:in `block in <top (required)>'
C:/Users/lexstarwind/.local/share/gem/ruby/3.0.0/gems/ceedling-0.31.1/bin/ceedling:332:in `<top (required)>'
C:/Users/lexstarwind/AppData/Local/Microsoft/WindowsApps/ceedling:25:in `load'
C:/Users/lexstarwind/AppData/Local/Microsoft/WindowsApps/ceedling:25:in `<main>'
Tasks: TOP => test:test/test_bit_manipulation.c

lexstarwind, 2023-11-27:
Is there a fix or any new info about this problem? I can no longer use the plugin.

numaru, 2023-11-27:
The error seems to happen when ruby try to get the modification time of the file `test:test/test_bit_manipulation.c` but I would expect ceedling to check for `test/test_bit_manipulation.c` instead.

In the logs we can see that the error is reported by a ruby version 3.0.0 while you said you were using ruby 3.0.6. Which version is used when you run ceedling in the terminal?

It's also unexpected for me to see that the ceedling executable is in `AppData/Local/Microsoft/WindowsApps`. How did you installed ceedling?

lexstarwind, 2023-11-27:
I installed ceedling using the gem command. gem install ceedling.
When I run ruby --version from the command line. I get the following output - ruby 3.0.6p216 (2023-03-30 revision 23a532679b) [x64-mingw32]

miebka, 2023-11-28:
> Strangely enough the tests all run from command line

If you are running the test from the command line I assume that you are using something like 

`ceedling test:test_<module>`
e.g.
```
PS C:\projects\tut101> ceedling test:test_helloWorld.c

Test 'test_helloWorld.c'
------------------------
Running test_helloWorld.out...

--------------------
IGNORED TEST SUMMARY
--------------------
[test_helloWorld.c]
  Test: test_helloWorld_NeedToImplement
  At line (17): "Need to Implement helloWorld"

--------------------
OVERALL TEST SUMMARY
--------------------
TESTED:  1
PASSED:  0
FAILED:  0
IGNORED: 1
```

But from the error messages the plugin uses 
`ceedling test:test/test_<module>`

you are getting the same error message if you are adding the path on the command line call. e.g.  
ceedling test:**test**/test_helloWorld.c

```
PS C:\projects\tut101> ceedling test:test/test_helloWorld.c
rake aborted!
Errno::EINVAL: Invalid argument @ rb_file_s_mtime - test:test/test_helloWorld.c
C:/Users/Michael/.local/share/gem/ruby/3.0.0/gems/ceedling-0.31.1/bin/ceedling:345:in `block in <top (required)>'
C:/Users/Michael/.local/share/gem/ruby/3.0.0/gems/ceedling-0.31.1/bin/ceedling:332:in `<top (required)>'
C:/Users/Michael/AppData/Local/Microsoft/WindowsApps/ceedling:25:in `load'
C:/Users/Michael/AppData/Local/Microsoft/WindowsApps/ceedling:25:in `<main>'
Tasks: TOP => test:test/test_helloWorld.c
(See full trace by running task with --trace)

--------------------
OVERALL TEST SUMMARY
--------------------

No tests executed.
```

I also added the used versions here since the gems are installed under ruby version 3.0.0 even if 3.0.6 is installed on windows.
```
PS C:\projects\tut101> ruby --version
ruby 3.0.6p216 (2023-03-30 revision 23a532679b) [x64-mingw32]
PS C:\projects\tut101> ceedling version
   Ceedling:: 0.31.1
      Unity:: 2.5.4
      CMock:: 2.5.4
 CException:: 1.3.3
```

As stated above I used the plugin for round about 2 years without any problems but on a fresh installation it fails. Strangely old installations from my colleagues are still working without any problems.

numaru, 2023-11-28:
When we run the `ceedling help` command, it's written that we should not give a path the the test, only the source file name.

```bash
numaru@tower:~/dev/temp_sensor$ ceedling help
# ...
ceedling summary                      # Execute plugin result summaries (no build triggering)
ceedling test:*                       # Run single test ([*] real test or source file name, no path)
ceedling test:all                     # Run all unit tests (also just 'test' works)
ceedling test:delta                   # Run tests for changed files
ceedling test:path[dir]               # Run tests whose test path contains [dir] or [dir] substring
ceedling test:pattern[regex]          # Run tests by matching regular expression pattern
# ...
```

Using paths was working but it's maybe safer to use only file names in the extension if the case of 2 test files with the same name is not handled by ceedling.

lexstarwind, 2023-11-28:
To run the tests from command line, I just use the ceedling command. No parameters. I can also run using ceedling test:all

lexstarwind, 2023-11-28:
PS D:\Development\PSI\H2Scan\PSI_Software\CeedlingCMock> ceedling

Test 'test_bit_manipulation.c'
------------------------------
Running test_bit_manipulation.out...

-----------
TEST OUTPUT
-----------
[test_bit_manipulation.c]
  - "Temperature is High"
  - "Temperature is Low"

--------------------
OVERALL TEST SUMMARY
--------------------
TESTED:  4
PASSED:  4
FAILED:  0
IGNORED: 0

prestonAdmiral, 2023-11-29:
Just want to comment to bump this message, I have two coworkers who are experiencing this issue using the same setup as I am although I'm not getting this issue.

closed 2023-11-29 by numaru

numaru, 2023-11-29:
Thank you @lexstarwind, @miebka and @prestonAdmiral for the report and additional info.
The issue should be fixed by the release v1.11.0. Feel free to re-open it if it does not.

PR #122 (2023-11-29): :bug: Use only the file name in test:{TEST_ID} commands 
It's written in the ceedling code and help command that only the file
name should be used, not a path.

fix #120
commit abdd9dcf (2023-11-29): :bug: Use only the file name in test:{TEST_ID} commands

It's written in the ceedling code and help command that only the file
name should be used, not a path.

fix #120