numaru/vscode-ceedling-test-adapter #127: Extension says xml_tests_report is not enabled even after it is enabled

AaronFontaine-DojoFive, 2023-12-12:
I am trying to get this to work natively in Windows 10. I am testing this with the temp sensor demo: `ceedling example temp_sensor`. Ceedling works from the command line. I can do `ceedling test:all` and see that all 47 tests pass. I have added `xml_tests_report` to the plugins list but the extension is still unhappy. I don't know how to debug this further.

VS Code: 1.85.0
Ceedling: 0.31.1
Ruby: 2.7.3p183

Contents of the project.yml file:
```
---

# Notes:
# Sample project C code is not presently written to produce a release artifact.
# As such, release build options are disabled.
# This sample, therefore, only demonstrates running a collection of unit tests.

:project:
  :use_exceptions: FALSE
  :use_test_preprocessor: TRUE
  :use_auxiliary_dependencies: TRUE
  :build_root: build
  # :release_build: TRUE
  :test_file_prefix: Test
  :which_ceedling: gem
  :ceedling_version: '?'

#:release_build:
#  :output: TempSensor.out
#  :use_assembly: FALSE

:environment: []

:extension:
  :executable: .out

:paths:
  :test:
    - +:test/**
    - -:test/support
  :source:
    - src/**
  :support:
    - test/support

:defines:
  :common: &common_defines []
  :test:
    - *common_defines
    - TEST
  :test_preprocess:
    - *common_defines
    - TEST

:cmock:
  :when_no_prototypes: :warn
  :enforce_strict_ordering: TRUE
  :plugins:
    - :ignore
  :treat_as:
    uint8:    HEX8
    uint16:   HEX16
    uint32:   UINT32
    int8:     INT8
    bool:     UINT8

:libraries:
  :system:
    - m

:plugins:
  :load_paths:
    - "#{Ceedling.load_path}"
  :enabled:
    - stdout_pretty_tests_report
    - xml_tests_report
    - module_generator
    - gcov
    - compile_commands_json
```

The XML report exists and is generated to `build/artifacts/test/report.xml`. If I delete the build directory and try to use the extension, I get this error without anything even being generated. It is somehow preemptively and incorrectly deciding it cannot run ceedling. If I run ceedling from the command line, the build directory is regenerated including the `report.xml` file.

Here is the error in VS Code.
[image]

JuPrgn, 2023-12-21:
Hope this can help : I had something similar and was also using gcov plugin. My gcov was configured with : 
```
:gcov:
  :xml_report: FALSE
```
Changing this to : 
```
:gcov:
  :xml_report: TRUE
```
Fixed the issue in my case. Maybe you can remove gcov to check ? I guess gcov default value is FALSE ?

AaronFontaine-DojoFive, 2024-01-02:
I got this to work. The trick is to add a section for the `xml_test_report` plugin that defines the location of the xml report artifact.

```
:xml_tests_report:
  :artifact_filename: report_xunit.xml
```

You need that in addition to listing `xml_tests_report` in the list of enabled plugins.

I didn't have to do anything in the section for the `:gcov:` plugin. I'm not sure how that it got it working for you.

numaru, 2024-01-02:
Thank you for contributing and posting a solution. The code which print the given error message is pretty simple.

https://github.com/numaru/vscode-ceedling-test-adapter/blob/4283fdf91f10a13a5208d8d8e67ea7070d3dd3a0/src/adapter.ts#L284-L293

I don't understand how adding a `:xml_tests_report:` section in `project.yml` fixed anything but I'm happy you found a workaround.