numaru/vscode-ceedling-test-adapter #134: Ceedling 0.32 is nearing release — changes to be aware of

mkarlesky, 2024-03-04:
@numaru Thank you so much for all your work on this extension. I am the original author of Ceedling from way back and rejoined the project starting with the big 0.32 effort. We've heard from many how indispensable your extension is. ❤️

As you know, Ceedling 0.32 has been in the works for a while now and includes some big changes. We're close to an official release. We're down to a few weeks worth of documentation updates, small code changes, etc. I checked in with an avid user of your VSCode extension to see where it was with the 1.12.0 release last month. He reported that it had been working well until some very recent 0.32 pre-release changes. He cited plugin-related issues. Indeed, the last big set of changes for 0.32 has been built-in plugin renaming, consolidation, and various updates.

There's a small amount of work left to do on [Ceedling 0.32 documentation](https://github.com/ThrowTheSwitch/Ceedling/tree/test/ceedling_0_32_rc/docs), both formatting and content, but it's nearly complete and is quite stable now. _BreakingChanges_ and _ReleaseNotes_ will be of greatest interest to you (I'm supposing you've already been looking at these). Do see the list of plugin name changes and related in the most recent documentation updates. The only thing not mentioned in the notes is that updates to the `dependencies` plugin are nearing completion, and it will most likely entirely subsume the `subprojects` plugin (i.e. `subprojects` will be going away).

Nothing significant is currently left on the to-do list before Ceedling 0.32 release. If you have any questions or needs, please just reach out.

simeon-s1, 2024-07-30:
Hello, i have reworked the extension to work with the new Ceedling release (backwards compatible). I opened a PR and wrote an email to @numaru. If there is no answer, everyone feel free to use my [fork](https://github.com/simeon-s1/vscode-ceedling-test-adapter/tree/ceedling-1.0.0-compatibility-and-multi-project-setup).

numaru, 2024-09-20:
As you might expect, I don't use ceedling and this extension anymore. But I still planned to do the work to support v1.0 when the release will be out. I would avoid to start working in advance because the release is "soon out" for so long. And since there is no defined machine interface between ceedling and this extension, it's only made of tiny hacks to automate the process, any harmless change in ceedling could break the extension. And it already happened in the past for the v0.32-pre.

I know some people started using the develop branch of ceedling for good reasons but we cannot guarantee that we won't break their projects by pushing an update later.

@mkarlesky I admire the efforts you guys put into the release v1.0 and I think ceedling deserve it. You can count on me to adapt and publish the extension when the release will be out. Even if it took me more than 6 months to answer to this message.

mkarlesky, 2024-10-02:
@numaru It's taken us way longer than we thought to finish the huge 1.0.0 release. Everything — new features, bug fixes, docs, deployment packaging — is done and ready for release. We're just running some stress tests on the new multi-threaded builds because of an issue report. As you can imagine, multi-threading is tricky to get right. As soon as we have confidence all is working correctly or some final threading bugs are fixed, we're finally pushing this thing out.

rossb93, 2025-01-14:
Great work on the release of ceedling 1.0.0 @mkarlesky, the tool is invaluable for me at my workplace and looking forward to using the new version. @numaru The test explorer tool also is hugely beneficial for development, I just wanted to ask if it is compatible with the new release of ceedling?

mkarlesky, 2025-01-14:
Thank you, @rossb93. We're steadily working away at a handful of bug fixes and improvements after 1.0.0 encountered the real world!

numaru, 2025-11-06:
@mkarlesky, seems like I missed the release date by a lot. I see you guys forked and published to the vscode store. I'm sure you will do better maintenance than me. Would you be okay with me marking this one as deprecated and redirecting to yours in the README.md? Maybe only for projects >=1.0.0 if you don't want to be bothered by legacy.