numaru/vscode-ceedling-test-adapter #137: Ceedling 0.32 - plugin xml_tests_report does not exist

awiegel, 2024-07-12:
[image]

The xml_tests_report plugin does not exist for the current ceedling 0.32 rc.

Related to #134

the-working-rene, 2024-07-15:
This seems to be related to the breaking changes for the upcoming release, see [Consolidation of test report generation plugins](https://github.com/ThrowTheSwitch/Ceedling/blob/test/ceedling_0_32_rc/docs/BreakingChanges.md#consolidation-of-test-report-generation-plugins-%EF%B8%8F-report_tests_log_factory).

Joel-Skarper, 2025-01-09:
Ceedling v1.0 has been released at last, and as announced by the RCs, it breaks the compatibility with the test adapter

act-ses, 2025-01-20:
does someone know if there is a workaround to solf this issue?

BuriKizilkaya, 2025-01-30:
Can the adapter be used with the ceedling version 1.0.0?

rmolinete, 2025-02-05:
It would be great to add support for ceedling V1

act-ses, 2025-02-05:
Ceedling V1.0.0 is supported by this fork:
https://github.com/simeon-s1/vscode-ceedling-test-adapter/

act-ses, 2025-02-06:
You have to pack the extension files into a vsix and install it. Make sure that you haven't istalled the "original" Ceedling Test Explorer.

https://code.visualstudio.com/api/working-with-extensions/publishing-extension

You can skip the step of publishing if you use it on your own.

rmolinete, 2025-02-06:
> Ceedling V1.0.0 is supported by this fork: https://github.com/simeon-s1/vscode-ceedling-test-adapter/

I've tried this fork and I can confirm it works for me with ceedling v1.0.0, thank you

MathieuBahin, 2025-02-07:
Hi, I tried to follow your instructions but I failed to make it work.

> You have to pack the extension files into a vsix and install it. Make sure that you haven't istalled the "original" Ceedling Test Explorer.
> 
> https://code.visualstudio.com/api/working-with-extensions/publishing-extension
> 
> You can skip the step of publishing if you use it on your own.

I installed npm ran vsce package.
First I had an issue with rimraf and esbuild so I installed them with npm i -g 
Now I have an issue with the vsce package command with yields this :
I tried to install xml2js, js-yaml, tree-kill znd async-mutex with npm i -g  but to no avail
Anyone as an idea ? Sorry I'm not a js / ts / npm user I do embedded software
```

 vsce package
Executing prepublish script 'npm run vscode:prepublish'...

> vscode-ceedling-test-adapter@1.12.0 vscode:prepublish
> npm run -S esbuild-base -- --minify

> vscode-ceedling-test-adapter@1.12.0 esbuild-base
> rimraf out && esbuild ./src/main.ts --bundle --outfile=out/src/main.js --external:vscode --format=cjs --platform=node --minify

X [ERROR] Could not resolve "vscode-test-adapter-api"

    src/main.ts:2:49:
      2 │ import { TestHub, testExplorerExtensionId } from 'vscode-test-adapter-api';
        ╵                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~

  You can mark the path "vscode-test-adapter-api" as external to exclude it from the bundle, which
  will remove this error and leave the unresolved path in the bundle.

X [ERROR] Could not resolve "vscode-test-adapter-util"

    src/main.ts:3:37:
      3 │ import { TestAdapterRegistrar } from 'vscode-test-adapter-util';
        ╵                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~

  You can mark the path "vscode-test-adapter-util" as external to exclude it from the bundle, which
  will remove this error and leave the unresolved path in the bundle.

X [ERROR] Could not resolve "async-mutex"

    src/adapter.ts:3:22:
      3 │ import { Mutex } from 'async-mutex';
        ╵                       ~~~~~~~~~~~~~

  You can mark the path "async-mutex" as external to exclude it from the bundle, which will remove
  this error and leave the unresolved path in the bundle.

X [ERROR] Could not resolve "tree-kill"

    src/adapter.ts:4:22:
      4 │ import tree_kill from 'tree-kill';
        ╵                       ~~~~~~~~~~~

  You can mark the path "tree-kill" as external to exclude it from the bundle, which will remove
  this error and leave the unresolved path in the bundle.

X [ERROR] Could not resolve "js-yaml"

    src/adapter.ts:7:17:
      7 │ import yaml from 'js-yaml';
        ╵                  ~~~~~~~~~

  You can mark the path "js-yaml" as external to exclude it from the bundle, which will remove this
  error and leave the unresolved path in the bundle.

X [ERROR] Could not resolve "xml2js"

    src/adapter.ts:8:19:
      8 │ import xml2js from 'xml2js';
        ╵                    ~~~~~~~~

  You can mark the path "xml2js" as external to exclude it from the bundle, which will remove this
  error and leave the unresolved path in the bundle.

6 of 8 errors shown (disable the message limit with --log-limit=0)
node:child_process:921
    throw err;
    ^

Error: Command failed: C:\Users\mbahin\AppData\Roaming\npm\node_modules\esbuild\node_modules\@esbuild\win32-x64\esbuild.exe ./src/main.ts --bundle --outfile=out/src/main.js --external:vscode --format=cjs --platform=node --minify
    at genericNodeError (node:internal/errors:983:15)
    at wrappedFn (node:internal/errors:537:14)
    at checkExecSyncError (node:child_process:882:11)
    at Object.execFileSync (node:child_process:918:15)
    at Object.<anonymous> (C:\Users\mbahin\AppData\Roaming\npm\node_modules\esbuild\bin\esbuild:221:28)
    at Module._compile (node:internal/modules/cjs/loader:1562:14)
    at Object..js (node:internal/modules/cjs/loader:1699:10)
    at Module.load (node:internal/modules/cjs/loader:1313:32)
    at Function._load (node:internal/modules/cjs/loader:1123:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14) {
  status: 1,
  signal: null,
  output: [ null, null, null ],
  pid: 34492,
  stdout: null,
  stderr: null
}

Node.js v22.13.1
 ERROR  npm failed with exit code 1
```

act-ses, 2025-02-07:
I am also more familiar with embedded software ;)

if i remember right i have installed vsec with `npx install -g @vscode/vsce`
i can't remember installed anything else.

however i re-packed it and this is shown (only the first lines copied)

```
vsce package
Executing prepublish script 'npm run vscode:prepublish'...

> vscode-ceedling-test-adapter-multi@2.0.0 vscode:prepublish
> npm run -S esbuild-base -- --minify

> vscode-ceedling-test-adapter-multi@2.0.0 esbuild-base
> rimraf out && esbuild ./src/main.ts --bundle --outfile=out/src/main.js --external:vscode --format=cjs --platform=node --minify

  out\src\main.js  201.7kb

Done in 47ms
....
```

by writing this lines I notice that you are trying to pack the 'original' version 1.12.0 which should work either. I have only packed the forked version (2.0.0) and it worked for me as you can see from the log above.

you may try it with the forked version and see if it works.

Tordek, 2025-03-03:
In order to build the fixed version you must:

1. Switch to the proper branch: `ceedling-1.0.0-compatibility-and-multi-project-setup`. The default branch doesn't have the fixes.
2. `npm i` to install dependencies
3. `npm run package`

Afterwards you can use the working version.